treehouses/builder #5: Communication over Bluetooth

dogi, 2017-06-07:


kanishk1010, 2017-07-27:
The following solution has been tested to work on MacOS and Linux (Ubuntu):

Modify the Raspbian image as you would generally do with the treehouse-builder:

```./treehouse-builder --chroot```

Create a file in the the /home/pi directory. Let's name it 'btserial.sh'.

```sudo nano /home/pi/btserial.sh```

Copy following script to your file:

```bash
#!/bin/bash -e

#Edit the display name of the RaspberryPi so you can distinguish
#your unit from others in the Bluetooth console
#(very useful in a class setting)

echo PRETTY_HOSTNAME=raspberrypi > /etc/machine-info

# Edit /lib/systemd/system/bluetooth.service to enable BT services
sudo sed -i: 's|^Exec.*toothd$| \
ExecStart=/usr/lib/bluetooth/bluetoothd -C \
ExecStartPost=/usr/bin/sdptool add SP \
ExecStartPost=/bin/hciconfig hci0 piscan \
|g' /lib/systemd/system/bluetooth.service

# create /etc/systemd/system/rfcomm.service to enable 
# the Bluetooth serial port from systemctl
sudo cat <<EOF | sudo tee /etc/systemd/system/rfcomm.service > /dev/null
[Unit]
Description=RFCOMM service
After=bluetooth.service
Requires=bluetooth.service

[Service]
ExecStart=/usr/bin/rfcomm watch hci0 1 getty rfcomm0 115200 vt100 -a pi

[Install]
WantedBy=multi-user.target
EOF

# enable the new rfcomm service
sudo systemctl enable rfcomm

# start the rfcomm service
sudo systemctl restart rfcomm
```

Save the file, and then make it executable by updating its permissions like so:

```chmod 755 /home/pi/btserial.sh```

Now you have the basics of the script required to turn on the Bluetooth service and configure it.  But to do this 100% headless, you’ll need to run this new script on startup. Let’s edit /etc/rc.local to launch this script automatically.

```sudo nano /etc/rc.local```

Add the following lines after the initial comments:

```bash
#Launch bluetooth service startup script /home/pi/btserial.sh
sudo /home/pi/btserial.sh &
```

Your Raspbian image is now configured to make your Raspberry Pi ready to be paired via bluetooth on boot. Exit the `chroot` environment and write the modified image to SD card. Boot the RPi with the SD card.

**To make a serial connection on MacOS:**

Open Bluetooth Preferences on your Mac and look for 'raspberrypi'. Once you find it, click on 'pair.'
Now, open a terminal window and check the name of the serial connection:
`ls /dev/cu.*`
This should produce a list of available serial ports, one of which should now be named after your Pi. If the name of the serial connection is 'cu.raspberrypi-SerialPort', we can connect to the Pi using:
`screen /dev/cu.raspberrypi-SerialPort 115200`  

**To make a serial connection on Ubuntu:**

Open Bluetooth Settings and click on '+' to look for a list of available devices. Once you find 'raspberrypi',  click on 'next' and complete the pairing process. Now click on 'raspberrypi' in the list of paired devices under Bluetooth settings and note the MAC ID.

Open Terminal now and enter the following to kill all 'rfcomm' connections, if already active:
`sudo killall rfcomm`
Create a serial Bluetooth connection with the Pi using the MAC ID:
`sudo rfcomm connect /dev/rfcomm0 <MAC ID> 1 &`
Now you can access the Terminal session on your Pi using 'Screen'
`sudo screen /dev/rfcomm0`
Note: Please make sure screen is installed. You can install it using:
`sudo apt-get install screen
`

kanishk1010, 2017-08-07:
The above procedure exposes the Raspberry Pi to be paired by any unverified devices. So, it is highly advisable to disable the Bluetooth discoverability and pairing as soon as you have connected to the Raspberry Pi device. 

Run following command on The RPi console:

```bash 
sudo bluetoothctl
```
This will take you to the 'bluetooth' console, which looks like this:

**[bluetooth]**#

Turn off discoverability and pairing using following commands:

**[bluetooth]**# `discoverable no`
**[bluetooth]**# `pairing no`

Exit bluetoothctl:

**[bluetooth]**# `quit`

Note: If you restart your computer, you would have to run these commands again.

xyb994, 2017-11-22:
Can confirm ran the `btserial.sh` script found in [the comment above](https://github.com/ole-vi/treehouse-builder/issues/5#issuecomment-318274730) once on my RPi, I was able to connect to the RPi through bluetooth following directions for Mac.

commit e7f86010 (2017-12-28): add bluetooth support (fixes #5) (fixes #159) (fixes #160) (#165)
closed 2017-12-28 by dogi