treehouses/builder #14: Raspberry pi Wifi Hotspot

leonardmensah, 2017-06-14:


paulbert, 2017-08-02:
Here's what I did to create a hotspot using the onboard WiFi as an access point and a USB wireless adapter to connect to the internet:

Followed [the official Raspberry Pi documentation](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md) up to `USING THE RASPBERRY PI AS AN ACCESS POINT TO SHARE AN INTERNET CONNECTION` to turn the onboard wifi into an access point

Then I followed the `Configure IP routing between the wireless and Ethernet` section of [this article](https://www.maketecheasier.com/set-up-raspberry-pi-as-wireless-access-point/), adjusting the IP routing commands to:

```
sudo iptables -t nat -A POSTROUTING -o wlan1 -j MASQUERADE
sudo iptables -A FORWARD -i wlan1 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i wlan0 -o wlan1 -j ACCEPT
```

Looking into getting this working with only the onboard wifi currently

Chris-Boe, 2017-08-02:
I've also gotten this working by forwarding the ethernet connection through the USB TP-Link Wi-Fi dongle that @dogi  included in the RPI package (eth0 -> wlan1), making the USB Wi-Fi dongle the Access Point.

The instructions were virtually the same as the official RPi documentation as Paul linked above, but with different IP routing commands:
`sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
sudo iptables -A FORWARD -i eth0 -o wlan1 -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i wlan1 -o eth0 -j ACCEPT`

Additionally, these routing settings can be saved using the iptables-persistent tool and the following command after the previous commands:
'sudo sh -c "iptables-save > /etc/iptables/rules.v4"
This saves settings so each reboot is the same.

paulbert, 2017-08-02:
I just got the onboard wifi card to work as an access point with internet on its own using the instructions found at this post: (https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=138730#p938306)

Didn't do it on a clean install, so maybe someone else could try following that post to confirm this works?

paulbert, 2017-08-09:
To have it all in an easier to read format:

1. `apt-get install hostapd dnsmasq`
2. Edit /etc/dnsmasq.conf to be:
```
interface=lo,uap0
no-dhcp-interface=lo,wlan0
dhcp-range=192.168.2.100,192.168.2.200,12h
```
3. Edit /etc/hostapd/hostapd.conf (need to choose what to do about passphrase)
```
interface=uap0
ssid=pipi
hw_mode=g
channel=6
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=0123456789
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```
4. Add the following to /etc/network/interfaces
```
auto uap0
  iface uap0 inet static
  address 192.168.2.1
  netmask 255.255.255.0
```
5. Create a file /usr/local/bin/hostapdstart
```
iw dev wlan0 interface add uap0 type __ap
service dnsmasq restart
sysctl net.ipv4.ip_forward=1
iptables -t nat -A POSTROUTING -s 192.168.2.0/24 ! -d 192.168.2.0/24 -j MASQUERADE
ifup uap0
hostapd /etc/hostapd/hostapd.conf
```
6. `chmod 667 /usr/local/bin/hostapdstart`
7. Add to /etc/rc.local
```
hostapdstart >1&
```

dogi, 2017-09-19:
todo:
- [ ] https://github.com/ole-vi/treehouse-builder/commit/05014c151195ed7c603dc3a5c966ac7f9b8316ea
- [ ] https://github.com/ole-vi/pirateship/commit/7353976aed050bab6a3f5c5619587fc0566a5cbd
- [ ] fix password less part

dogi, 2017-09-20:
In issue https://github.com/ole-vi/pirateship/issues/16 I described how config file has to be made for password less hotspot

commit dd0f99bb (2017-09-22): update pirateship to 0.2.3 (fixes #14)(#81)
closed 2017-09-22 by dogi