treehouses/builder #17: create a new rpi-couchdb docker iamge

dogi, 2017-06-14:


mappuji, 2017-08-10:
Last effort:
github.com/mappuji/rpi-couchdb

Todo:
[ ] Makes alien RPi works in MAC —> Treehouses
[ ] Entrypoint
[ ] Too much packages (X11)
[ ] Clean up the [repository](github.com/mappuji/rpi-couchdb) and transfer to `ole-vi`

mappuji, 2017-08-12:
After hacking session on Saturday August 12:

- Alien RPi is not working natively in Docker in MacOS
- We already fix log setting in rpi-couchdb `rpi-couchdb:beta2`

Todo:
- [ ] Create vagrantfile for RPi docker creation
- [ ] Fix Entrypoint
- [ ] Reduce size of `rpi-couchdb` docker by removing X11 packages
- [ ] Cleanup the repository and transfer to `ole-vi`

mappuji, 2017-08-16:
After hacking session on Saturday August 15:

- [x] Test vagrantfile for RPi docker creation
https://github.com/ole-vi/octopirate
- [x] Fix Entrypoint
- [ ] Reduce size of rpi-couchdb docker by removing X11 packages
- [ ] Cleanup the repository and transfer to ole-vi

mappuji, 2017-08-17:
After hacking session on Saturday August 15:

- [x] Test vagrantfile for RPi docker creation
https://github.com/ole-vi/octopirate
- [x] Fix Entrypoint
- [ ] Reduce size of rpi-couchdb docker by removing X11 packages
- [x] Cleanup the repository and transfer to ole-vi

closed 2017-08-18 by mappuji