treehouses/builder #97: raspbian stretch is out

dogi, 2017-10-05:


i5o, 2017-10-12:
- [ ] default
- [ ] rename <hostname>
- [ ] wifi <ESSID> [password]
- [ ] staticwifi <ip> <mask> <gateway> <dns> 
- [ ] ethernet <ip> <mask> <gateway> <dns>
- [ ] hotspot <ESSID> [password]  
- [ ] expandfs
- [ ] detectrpi 
- [ ] version 
- [ ] docker 
- [ ] password <password>

i5o, 2017-10-12:
```
root@raspberrypi:/home/pi# pirateship rename ignaciopi
/usr/lib/node_modules/@pirateship/cli/lib/ReconfigureHostname.js:34
            if (error) return callback(error)
                              ^

TypeError: callback is not a function
    at /usr/lib/node_modules/@pirateship/cli/lib/ReconfigureHostname.js:34:31
    at ChildProcess.exithandler (child_process.js:278:5)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at maybeClose (internal/child_process.js:927:16)
    at Socket.stream.socket.on (internal/child_process.js:348:11)
    at emitOne (events.js:115:13)
    at Socket.emit (events.js:210:7)
    at Pipe._handle.close [as _onclose] (net.js:547:12)
root@raspberrypi:/home/pi#
```

i5o, 2017-10-12:
Testing in rpi2

```
root@raspberrypi:/home/pi# pirateship hotspot Ignaxio 12345678
This pirateship has anchored successfully!
exec error: Error: Command failed: ifdown uap0
ifdown: unknown interface uap0

exec error: Error: Command failed: ifup uap0
ifup: unknown interface uap0

root@raspberrypi:/home/pi#
```

i5o, 2017-10-12:
pirateship hostname

Funny thing, the hostname is there even after the commadn failed

i5o, 2017-10-18:
**--- retesting in raspberry pi 3 ---**

i5o, 2017-10-18:
pirateship
- [ ] default
- [ ] rename <hostname>
- [ ] wifi <ESSID> [password]
- [ ] staticwifi <ip> <mask> <gateway> <dns> 
- [ ] ethernet <ip> <mask> <gateway> <dns>
- [ ] hotspot <ESSID> [password]  
- [ ] expandfs
- [ ] detectrpi 
- [ ] version 
- [ ] docker 
- [ ] password <password>

i5o, 2017-10-18:
```
root@raspberrypi:/home/pi# pirateship rename ignaciopi
/usr/lib/node_modules/@pirateship/cli/lib/ReconfigureHostname.js:34
            if (error) return callback(error)
                              ^

TypeError: callback is not a function
    at /usr/lib/node_modules/@pirateship/cli/lib/ReconfigureHostname.js:34:31
    at ChildProcess.exithandler (child_process.js:278:5)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at maybeClose (internal/child_process.js:927:16)
    at Socket.stream.socket.on (internal/child_process.js:348:11)
    at emitOne (events.js:115:13)
    at Socket.emit (events.js:210:7)
    at Pipe._handle.close [as _onclose] (net.js:547:12)
root@raspberrypi:/home/pi# hostname
raspberrypi
root@raspberrypi:/home/pi# exit
```

i5o, 2017-10-18:
**Action: will fix hostname and continue**

i5o, 2017-10-19:
```
root@ignaciopi:/home/pi# pirateship password ignacio
password change success
root@ignaciopi:/home/pi#
```

commit b1af50e2 (2017-11-24): switch to raspbian stretch (fixes #97) (#118)
closed 2017-11-24 by dogi