treehouses/builder #147: debian image in vagrant box doesn't support 'overlay2' in docker

i5o, 2017-12-14:
stretch must be used to make it work
https://docs.docker.com/engine/userguide/storagedriver/selectadriver/#docker-ce

dogi, 2017-12-19:
https://github.com/ole-vi/ole--vagrant-treehouses/pull/6
https://github.com/ole-vi/ole--vagrant-treehouses/commit/729374ea5a5e5f646f77ab0ca8b6938ad53c5ee4

@i5o can you confirm that it works now?

i5o, 2017-12-20:
vagrant@treehouses:~$ docker info
Containers: 1
 Running: 0
 Paused: 0
 Stopped: 1
Images: 2
Server Version: 17.09.1-ce
Storage Driver: overlay2
 Backing Filesystem: extfs
 Supports d_type: true
 Native Overlay Diff: true
Logging Driver: json-file
Cgroup Driver: cgroupfs
Plugins:
 Volume: local
 Network: bridge host macvlan null overlay
 Log: awslogs fluentd gcplogs gelf journald json-file logentries splunk syslog
Swarm: inactive
Runtimes: runc
Default Runtime: runc
Init Binary: docker-init
containerd version: 06b9cb35161009dcb7123345749fef02f7cea8e0
runc version: 3f2f8b84a77f73d38244dd690525642a72156c64
init version: 949e6fa
Security Options:
 seccomp
  Profile: default
Kernel Version: 4.9.0-4-amd64
Operating System: Debian GNU/Linux 9 (stretch)
OSType: linux
Architecture: x86_64
CPUs: 1
Total Memory: 2.124GiB
Name: treehouses
ID: BMBF:AMGA:N24F:7Q3E:6OMP:JUXE:5YFT:CBLN:52H3:IBJL:XWYT:LZPL
Docker Root Dir: /var/lib/docker
Debug Mode (client): false
Debug Mode (server): false
Registry: https://index.docker.io/v1/
Experimental: false
Insecure Registries:
 127.0.0.0/8
Live Restore Enabled: false

WARNING: No swap limit support
vagrant@treehouses:~$ 

fixed

closed 2017-12-20 by i5o