treehouses/builder #256: getting ssh keys somtimes from github fails

i5o, 2018-04-12:
Gettings key from: https://api.github.com/orgs/open-learning-exchange/members?per_page=20&page=1
jq: error (at <stdin>:4): Cannot index string with string "url"

https://travis-ci.org/treehouses/builder/builds/365371207#L2774

i5o, 2018-04-12:
```
{
  "message": "API rate limit exceeded for 1.2.3.4. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)",
  "documentation_url": "https://developer.github.com/v3/#rate-limiting"
}
ignacio@kuckuck:~$ page_number=1
ignacio@kuckuck:~$     api_url="https://api.github.com/orgs/open-learning-exchange/members?per_page=20&page=$page_number"
ignacio@kuckuck:~$ $(curl -s "$api_url"| jq '.[].url' | sed -rn "s/(.*)\\/users\\/(.*)\"/https:\\/\\/github.com\\/\\2.keys/p")
jq: error (at <stdin>:4): Cannot index string with string "url"
ignacio@server:~$ 
```

dogi, 2018-04-14:
add test on the to see if failed

i5o, 2018-04-14:
```
for i in $(seq 1 100); do echo $(curl -s "$api_url"| jq '.[].url' | sed -rn "s/(.*)\\/users\\/(.*)\"/https:\\/\\/github.com\\/\\2.keys/p"); done
```

how to reproduce

PR #258 (2018-05-03): check authorized_keys (fixes #256)
Fixes #256 

Is it better to check `$keys` in `30_ssh_keys.sh` and not generate `authorized_keys` at all if something went wrong? It would help keep the addition of lines to `treehouse-builder` minimal too.
commit a61b3eb3 (2018-05-03): check authorized_keys file length (fixes #256) (#258)
closed 2018-05-03 by dogi