treehouses/builder #417: usb mobile internet dongle

dogi, 2019-02-21:


dogi, 2019-02-28:
https://github.com/ascoderu/opwen-cloudserver

dogi, 2019-03-20:
Clemens wrote us
```
From a high level point of view, the modem internet flow is like this:

1) Confirm modem is plugged in via lsusb
2) Confirm if the modem is in modem mode or storage mode via lsusb. If the modem is in storage mode, use usb_modeswitch to turn it into modem mode.
3) Use wvdial to create the dial-up connection.
4) Use the internet as normal.
5) Terminate the wvdial process to stop the internet connection.

For code pointers, you can take a look at this class, specifically the __call__ method: https://github.com/ascoderu/opwen-webapp/blob/f9619e99fe07105ef9058bcfa7a7e65f1fc99f15/opwen_email_client/webapp/actions.py#L76-L148

I haven't used your specific modem, but you can find the type of information you'll need for mode switching it here: https://github.com/ascoderu/opwen-webapp/blob/f9619e99fe07105ef9058bcfa7a7e65f1fc99f15/opwen_email_client/domain/modem/e353.py

There's an apt package called usb-modeswitch-data that may contain the correct target mode and magic bytes to switch your modem, if not, you'll have to do a lot of painful internet research.

You can also use a modem that doesn't require mode-switching such as this one: https://consumer.huawei.com/solutions/m2m-solutions/en/products/tech-specs/ms2131-en.htm

In order to establish the internet connection via the modem, you'll need dial-up information for your SIM card provider in wvdial configuration format. The wvdial tool then takes care of the heavy lifting of creating the requisite network interfaces, starting ppp, etc. You can see an example of what the configuration may look like here: https://github.com/ascoderu/opwen-webapp/blob/f9619e99fe07105ef9058bcfa7a7e65f1fc99f15/opwen_email_client/domain/sim/hologram.py
Hope that helps.
```

i5o, 2019-03-21:
FWIW our usb modem is plug & play, so basically we just plug it and a new ethX interface gets created

dogi, 2019-07-03:
@i5o thanks

closed 2019-07-03 by dogi