treehouses/builder #452: Missing icons/images on desktop

sc4773rm1nd, 2019-07-03:
See title.
[image]

dogi, 2019-07-12:
https://www.raspberrypi.org/forums/viewtopic.php?t=86736

@sc4773rm1nd thanks for this link

dogi, 2019-07-12:
potential packages
```
libraspberrypi-bin
libraspberrypi-dev
libraspberrypi-doc
libraspberrypi0
lxappearance-obconf
lxpanel-data
lxpanel
lxplug-bluetooth
obconf
pi-greeter
pipanel
raspberrypi-ui-mods
raspi-config
rc-gui
rp-prefapps
xdg-utils
```

sc4773rm1nd, 2019-07-12:
Even after the fix in the above link, problem not fixed

dogi, 2019-07-13:
https://gitter.im/open-learning-exchange/raspberrypi?at=5d2a4eff7dbfff3242a1e787

```
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for shared-mime-info (1.10-1) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for gnome-menus (3.31.4-3) ...
```

commit ecbc0aa0 (2019-07-14): not breaking GUI icons and background (fixes #452) (#468)
closed 2019-07-14 by dogi