treehouses/builder #616: investigate and fix or escalate why version 0.23 of pybluez is breaking

dogi, 2020-01-06:
> ```
> root@treehouses:~# service rpibluetooth status
> ● rpibluetooth.service - Bluetooth server
>    Loaded: loaded (/etc/systemd/system/rpibluetooth.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Thu 2019-09-26 00:31:17 UTC; 3 months 11 days ago
>   Process: 513 ExecStart=/usr/bin/python3 /usr/local/bin/bluetooth-server.py & (code=exited, status=1/FAILURE)
>  Main PID: 513 (code=exited, status=1/FAILURE)
> 
> Sep 26 00:31:17 treehouses python3[513]: Traceback (most recent call last):
> Sep 26 00:31:17 treehouses python3[513]:   File "/usr/local/bin/bluetooth-server.py", line 179, in <module>
> Sep 26 00:31:17 treehouses python3[513]:     server.run()
> Sep 26 00:31:17 treehouses python3[513]:   File "/usr/local/bin/bluetooth-server.py", line 77, in run
> Sep 26 00:31:17 treehouses python3[513]:     profiles=[bluetooth.SERIAL_PORT_PROFILE])
> Sep 26 00:31:17 treehouses python3[513]:   File "/usr/local/lib/python3.7/dist-packages/bluetooth/bluez.py", line 275, in advertise_service
> Sep 26 00:31:17 treehouses python3[513]:     raise BluetoothError (*e.args)
> Sep 26 00:31:17 treehouses python3[513]: bluetooth.btcommon.BluetoothError: no advertisable device
> Sep 26 00:31:17 treehouses systemd[1]: rpibluetooth.service: Main process exited, code=exited, status=1/FAILURE
> Sep 26 00:31:17 treehouses systemd[1]: rpibluetooth.service: Failed with result 'exit-code'.
> root@treehouses:~# treehouses image
> release-117
> ```
> this is release 117

dogi, 2020-01-06:
https://github.com/treehouses/builder/blob/master/scripts.d/30_bluetooth.sh#L10

kjong, 2020-01-06:
https://github.com/pybluez/pybluez

https://github.com/pybluez/pybluez/blob/master/CHANGELOG
Changelog shows nearly 4 years of changes between version 0.22 and 0.23

sente, 2020-01-07:
I think this is a potential solution:
https://www.raspberrypi.org/forums/viewtopic.php?t=133263

commit 5473f4cf (2020-04-09): update 30_bluetooth.sh (fixes #616) (#667)
closed 2020-04-09 by dogi