treehouses/builder #622: add package sl

dogi, 2020-01-13:
[image]

you may have to add enhance the PATH

dogi, 2020-01-16:
```
$ which sl
/usr/games/sl
```

jamjakpa, 2020-01-17:
> ```
> $ which sl
> /usr/games/sl
> ```

jamjakpa, 2020-01-17:
> ```
> $ which sl
> /usr/games/sl
> ```

@dogi can you give me more hints please

jamjakpa, 2020-01-17:
I am stuck

dogi, 2020-01-19:
`ln -sr /usr/games/sl /usr/local/bin/sl`

jamjakpa, 2020-01-20:
[image]

jamjakpa, 2020-01-20:
[image]

jamjakpa, 2020-01-20:
[image]

jamjakpa, 2020-01-20:
[image]

jamjakpa, 2020-01-20:
the filename is `32_sl.sh` <----create filename name of the scripts.

jamjakpa, 2020-01-21:
> `ln -sr /usr/games/sl /usr/local/bin/sl`

hi @dogi, I have created a new branch 'jamjakpa-image-sl1' , in that branch I have build the script and called it 32_sl.sh . I have also created a Release name in Gitbash: 
git pull 
git tag 
git tag release-image-sl1 

somehow i can't see the tag ? in builder

commit 150aa218 (2020-02-20): sl package added (fixes #622) (#624)
closed 2020-02-20 by dogi