treehouses/builder #635: Rpi3B+ wifi experimental image

Nida8930, 2020-02-10:
#### Experimental Image

Download image [here.](http://download.treehouses.io/experiment/treehouse-raspbiantwenty0.img.gz)

#### Overview

Using Rpi3B+

When attempting to connect the wifi using the image linked above the pi repeatedly failed to connect to the network. I found that on this new image the wifi country must be set to successfully connect to wifi.

#### Proposed Solution

By connecting to the pi via console cable I was able to run `sudo raspi-config`, navigate to Localization Options then Change Wi-fi Country and select United States. After selecting a wifi country I was able to use the `treehouses wifi` command as usual.

Nida8930, 2020-02-12:
Raspi-config does not work until you run treehouses default network and reboot. After this you are able to set the wifi country using raspi-config and then use treehouses wifi as normal.

`treehouses default network` does reset the wifi country. Once the wifi country is initially set by raspi-config everything still works even if it is reset later by `default network`.

I thought that maybe we could modify the template used for the wpa_supplicant.conf during `treehouses default network` to set the wifi country preemptively, but it appears that wifi is only unlocked when using raspi-config.

kjong, 2020-02-13:
It's difficult for me to test this as thoroughly as I'd like to since I can only connect to my raspberry pi through wifi (which obviously makes testing wifi related issues difficult).

However, I flashed the new image and after adding my wifi settings (name and password) in my autorunonce file and including the line `rfkill unblock wifi`, my raspberry pi had no problems connecting to wifi.

If you look at the [release notes](http://downloads.raspberrypi.org/raspbian/release_notes.txt) for the raspbian images, even though it's not under the latest release (it's under 2019-06-20), you have these lines:
```
* On-board 5GHz WiFi blocked by rfkill by default
    The block is removed when taking one of the following actions:
    - Selecting a locale in the first run wizard
    - Setting the WiFi country in the Raspberry Pi Configuration tool or the Network Settings applet
    - Setting the WiFi country in raspi-config
    - Providing a wpa_supplicant.conf file through the boot partition
    - Running 'rfkill unblock wifi'
```

Hopefully this is relevant and helpful.

LordJashin32, 2020-02-13:
I think the easiest solution is the one @kjong proposed with the autorunonce file.

dogi, 2020-02-13:
> `treehouses` `66........62` `1.13.4` `RPI4B` `release-raspbiantwenty0  `:
> y2gos5tkppeve.............emhawvi4fzlz6dad.onion
> 2200:2200 80:80 22:22
> `2020-02-13 01:54:53 UTC` wifi

Nida8930, 2020-02-13:
@kjong Perfect, thank you.

commit 17d22846 (2020-02-13): modify autorunonce file (fixes #635)
closed 2020-02-20 by dogi